Helder1981/OneForAll #2: Fix Readme

jinlucky, 2018-11-02:
Your Readme file is more advertising, compliments and pictures than instructions.

### 建议
* use markdown lists for features
   * use *w/* with consistancy
* use code block for code instructions
* use markdown syntax properly

#### 功能示例
- battery meter
- charging indicator
- wifi signal
- bluetooth indicator
- analog stick support
- built in controls
- volume

#### 代码块
1. explain to audience why you need *sudo*
2. explain to audience why you need *read, write and execute* on your install script

```shell
$ sudo su
```

```shell
# since you are in sudo from above step
git clone --recursive https://github.com/Helder1981/OneForAll.git \
cd OneForAll \
chmod a+rwx /home/pi/OneForAll/install.sh \
./install.sh
```

##### general
please tell people to **NOT** change GitHub clone directory name, you code has bad dependency on your folder structure being named *OneForAll*. people rename folders all the time using *git clone*

ie: git clone [path to repo] [folder name]
ie: git clone --recursive https://github.com/Helder1981/OneForAll.git CoolBatteryThing

You should build your code using $PWD and directory services

祝你有美好的一天

withgallantry, 2018-11-02:
Thanks for the comprehensive list. It is advertising I suppose, for the
features it has, but it does need more instructions.
We initially wrote this for ourselves and thought people may want to use
it, hence why the documentation is light. I fully take on board though that
it does need more documentation.
I’ve been working on an improved version so along with that release I will
add some instructions as per above.
With regards to hard dependency on the directories. This has already been
removed. I was sure this version didn’t have a hard dependence on directory
structure. I will check, however the newer version does not so it won’t be
an issue.
Thanks again for the feedback, I’ll revise the next readme

withgallantry, 2018-11-02:
I've deleted my previous post, I wrote it on mobile so it was poorly formatted.

This software was created specifically for use with the RetroPSU board but can be used with an ADS1015. It was a joint effort between a few people. This is why the documentation is focused on the RetroPSU and the installation instructions simple. The documentation is sufficient. I'm also sorry you feel that paying compliments to people who helped in the joint effort isn't worthy of being noted on the Readme. I do.

In regards to hard dependencies, this is only true for the installer. The actual OSD can used from any directory. We created an installer that uses a specific directory because it's easier for us to diagnose if users have problems if they're all configured the same. For this reason, we have used a hard coded directory. I will add a note not to change the directory if using the installer.

We will reopen this if users raise similar questions however so far we have not had any issues.
Also to note, it uses sudo because it needs access to uinput (for the controls) which requires elevated privileges.

closed 2018-11-02 by withgallantry