Helder1981/OneForAll #3: Icon size for composite screens

mrdink, 2018-12-13:
Spoke with @Helder1981 about this - it seems the icons show really tiny on composite screens. Not sure if it's fixable or not but might be worth looking into.

[image]

mrdink, 2018-12-13:
Also, just to give a little more info.

This is running latest retropie 4.4 with latest OneForAll script. Here's how my config.txt looks.

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=-8
overscan_right=-16
overscan_top=-24
#overscan_bottom=16

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1
dtparam=i2c_arm=on

# Disable Pi Status LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on

# Disable Rainbow Splash
disable_splash=1

# Overclocking
arm_freq=1000
gpu_freq=500
core_freq=500
sdram_freq=500
sdram_schmoo=0x02000020
over_voltage=2
sdram_over_voltage=2

# Rotate display
display_rotate=2
dtoverlay=hifiberry-dac
dtoverlay=i2s-mmap
```

Unboundclassic, 2019-05-16:
I have the same issue. Any more info guys?

withgallantry, 2019-05-16:
I have been working on allowing resolution size to be set but it's a fundamental change to the code so it has been an on and off project. I'll let you know when it's completed.

Unboundclassic, 2019-05-16:
I'd love to see this happen for sure. Right now on a 320x240 composite screen the battery level indicator is barely usable and the other icons are so small that their "states" are pretty much indistinguishable from each other. 

I'd prefer a "simple" command line option in the configuration file where maybe you have:

OSD SIZE = 3
(Choose 1-5. 1 being the smallest, 5 being the largest)

Then it just scales by 20% or something for each setting. I'm not sure what would be harder. An automatic scaling system seems like it would be harder to implement than just adding a scalable option. But I'm no expert. The scalable size option seems like the most flexible option to give it the most utility for the most people using different sized screens.

withgallantry, 2019-05-16:
Oh I wasn't aware the resolution was 320x240. It might be a case that the resolution is actually higher and being downscaled for the composite screen. 
I don't know if this works for composite but you could try forcing the resolution, it should also make the screen sharper in general:
```
hdmi_force_hotplug=1
hdmi_cvt=320 240 60 1 0 0 0
hdmi_group=2
hdmi_mode=87
display_rotate=2
```

Let me know if that works for you. If not I think your suggestion is a good one and I'll start implementing it this way.

Unboundclassic, 2019-05-16:
I don't believe any of those are applicable here as the HDMI settings in the .config file are ignored when outputting to composite. As far as I've seen in the Retropie documentation it's either one or the other. The HDMI settings only apply to the source connected to the HDMI port. Where composite screens are wired directly to the analogue output on the bottom of the Pi board.

Unboundclassic, 2019-05-17:
On a very related note, if the resizing could apply to the HUD as well that would be helpful. The small print on the HUD is hard to read at this resolution.

withgallantry, 2019-05-17:
I thought that might be the case, just wasn't sure as I've not worked with composite much. I've got a composite screen here so I will get a test rig going and get it resizing.

Unboundclassic, 2019-05-17:
For your reference I'd recommend these settings for composite in the config.txt file. Should give you 240P:

framebuffer_width=320
framebuffer_height=240
sdtv_mode=16
sdtv_progressive_scan=1
audio_pwm_mode=0