Helder1981/OneForAll #4: Overlay not showing when changing GPIO

patriziosotgiu, 2019-05-16:
I'm having an issue with the overlay not showing when setting the GPIO to a pin that doesn't equal to one of the other buttons. 

Examples: 
- If I set the `HOTKEY` to be the same as the `SELECT` button (5), everything works fine. 

- If I set the `HOTKEY` to a dedicated pin, in my case (13), I get a key error and the overlay doesn't show up. 
In this specific case, I can still change the volume and activate/deactivate bluetooth and wifi, so I know that the pin is right. 

Here's the error: 
```
May 15 00:07:54 retropie-gameboy-spw rc.local[295]: Traceback (most recent call last):
May 15 00:07:54 retropie-gameboy-spw rc.local[295]:   File "monitor_evdev.py", line 210, in handle_button
May 15 00:07:54 retropie-gameboy-spw rc.local[295]: KeyError: 13
```
and here's my `keys.cfg`

```
[GENERAL]
SHUTDOWN_DETECT = -1

[KEYS]
LEFT = 23
RIGHT = 0
DOWN = 1
UP = 22
BUTTON_A = 16
BUTTON_B = 6
BUTTON_X = 17
BUTTON_Y = 26
BUTTON_L1 = 27
BUTTON_R1 = 20
SELECT = 5
START = 12
HOTKEY = 13

[JOYSTICK]
ON_BY_DEFAULT=False
DISABLED=True
DEADZONE=300
VCC=1600

[BATTERY]
ENABLED=True
FULL_BATT_VOLTAGE=373
BATT_LOW_VOLTAGE=340
BATT_SHUTDOWN_VOLT=328
```

I spoke with someone else with my same build that said is working fine for him.
Could it be a bug that got recently introduced?

withgallantry, 2019-05-16:
It should be fine, there was a fix done a month ago to address this. I will do some more testing tonight and get to you though.

withgallantry, 2019-05-18:
So I've managed to create one scenario where this might happen and I'll get a bug fix done. Is the issue still occurring for you?

patriziosotgiu, 2019-05-18:
Yes, still happening on my end. A bug fix would be great! Thanks so much!

withgallantry, 2019-08-08:
This has finally been fixed in the new version as it's had a rewrite to optimise and add a few features in:
https://www.bluuplabs.com/oneforall-v2/

Once a few people have real world tested it I will be compiling and adding the binary to this repository. Currently the update is only on my source repository