Helder1981/OneForAll #5: Hot Key Not Registering/Operating Correctly

Unboundclassic, 2019-05-16:
I can't seem to get ES to recognize the Hot Key button. I've switched out both wires, gave it a new ground point, switched out buttons, and tried re-assigning it and switching to a different GPIO pin. No luck. I tested also both buttons. They both are working when connected, at least according to the voltage test. All the other buttons and analogue stick work great. In the Configure Input workflow I can assign all my other custom buttons properly but I get to the separate Hot Key button and nothing. 

Another oddity, the on screen display is on constantly from the moment ES boots up. From what I understand it should only appear when the button is pressed. In my case it's staying on always (which I kind of prefer to be honest) and it is not responding to the button at all. This makes me believe that it actually knows the button is there, it just won't let me use it. The OSD only showed up once I connect the button. 

I wonder if it installed incorrectly? If there a clean way of uninstalling it so I could try a fresh install? It was installed in the last week so it should be up to date. For further reference here is my current keys.cfg setup:

[image]

withgallantry, 2019-05-16:
Thanks for taking the time to write this up.
I think there might be some confusion due to my naming convention. Hotkey is actually the key to show the full screen overlay, it won't be detected by ES at all. If you'd like extra keys adding that is something I can certainly do. I might rename this HOTKEY to DISPLAY_HOTKEY instead to avoid confusion. The HOTKEY can be assigned the same pin number as any of the other buttons or a new GPIO pin for a stand alone button. 

The on screen display will always show, the software was adapted separately for the MintyPi which is where you may have read that the OSD will show only when the hotkey is pressed.

I hope that clears things up.

Unboundclassic, 2019-05-16:
Ah, OK. Helder told me it should be temporary and not persistent. And I think the verbiage in the "read me" makes it seem that way too. I tried assigning it to select and it worked there. I would definitely prefer to have the option to have it be it's own button as there are a lot other uses/needs for Select and I'd rather the HUD not pop up every time I push select. If you want it to be part of another key I would have it listed as DISPLAY_TOGGLE or OSD_TOGGLE or some other term and avoid HOTKEY at all as people will associate it with the Emulation Station Hotkey. 

Personal though the ideal for me would be something that works for both. I'd love a dedicated Hotkey button for the OSD and the same button also lets me do the system Hotkeys like saving and loading games and such.

So perhaps a something formatted like this:

HOTKEY ON SELECT BUTTON = 6
(Set to FALSE if using a separate button)

HOTKEY AS SEPARATE BUTTON = 13
(Set to FALSE if using the select button)

withgallantry, 2019-05-17:
I might leave hotkey in, but set it to -1 if not needed and then add OSD_TOGGLE which can also be a separate key. Does that sound like a viable solution?

Unboundclassic, 2019-05-17:
Sounds good. As long as it's added to the "what does what" list and the Read Me that should work.

withgallantry, 2019-08-08:
Finally got around to doing a rewrite of the software. The source is still a little hard to read as it's been optimised for speed (hence the one long file, which goes against best coding practices really). The info can be found here:
https://www.bluuplabs.com/oneforall-v2/

Have a look through it, let me know what you think. I've renamed a lot of the configuration and separated it out into two files. One for general configuration and one for the keys configuration. This should make everything a lot clearer.

Once a few people have real world tested it I will be compiling and adding the binary to this repository. Currently the update is only on my source repository

Unboundclassic, 2020-05-10:
@withgallantry how do I update it? It's been a while since I've been in Command line. Trying the original install commands again it pretty much just states that it's already there. Unless I need to uninstall this and reinstall? What's the best approach?

Unboundclassic, 2020-05-26:
Issue fixed with the newer version for here. Though they're still updating the install procedure: https://github.com/withgallantry/OneForAll

closed 2020-05-26 by Unboundclassic