Helder1981/OneForAll #7: Analog Stick Not Registering in Riecast/Dreamcast Correctly

Unboundclassic, 2019-06-03:
The PSP analog stick I have connected through the RetroPSU (appears to be the same one on Helder's site) is working perfectly everywhere with ES and other systems, except with Dreamcast. I go to configure the OneForAll controller for Riecast and all the other buttons register properly except the analog stick. It asks to press it all the way left and then all the way up. It won't register unless I press it into the diagonals or make a circle motion. Either way it won't configure properly which makes it more or less useless in games. I've seen it do this on two separate Retropie installs on different cards. Thoughts?

withgallantry, 2019-06-03:
Interesting. I've not heard of this before but I will investigate. Are these fresh installs of Retropie 4.4? 
I'm currently adding some features requested in other issues but I'll look at this whilst testing.

djobbydjobba, 2019-06-04:
Hi. It is not just related to this board. Teensy users can't register the analog stick either. This is a well known issue.

withgallantry, 2019-06-05:
Thanks for the additional info, I tested last night and indeed had the same issue. I will see what I can come up with although it may not be possible but I like to think there are alway solutions. I'll keep this issue updated.

Unboundclassic, 2019-06-27:
Is the more of a Riecast issue then? Definitely kills the playability of a bunch of games. Let us know if you find something. Thanks.

withgallantry, 2019-08-08:
I've updated the joystick handling functions so this should now work. You can find an example in the keysGamepad.cfg of the configuration needed for a Gamepad with a joystick. The info and the source repository can be found in the update post here:
https://www.bluuplabs.com/oneforall-v2/

Once a few people have real world tested it I will be compiling and adding the binary to this repository. Currently the update is only on my source repository