Helder1981/OneForAll #9: Gpio pinout for 6 buttons

tccabeza, 2019-06-23:
Im new at programing for pi 3 and i cant seem to understand the relation between the numbers assigned to each key in the keys file to each pinout of the gpio in the image attached in the readme.  I would like to know it if possible because i want to  add another 2 buttons for L2 and R2 and maybe two new front buttons for some arcade action. 

Thanks !

withgallantry, 2019-08-08:
Sorry for the long delay, the adding of new pins really made me think about what direction to take the OneForAll software in and I'm please to announce that the source has been updated. You can find the info here:
https://www.bluuplabs.com/oneforall-v2/
Have a look through it, let me know what you think. The key part is that you can now map as many new buttons as you like. 
Once a few people have real world tested it I will be compiling and adding the binary to this repository. Currently the update is only on my source repository.

withgallantry, 2020-01-09:
I'm going to close this, please re-open if needed.

closed 2020-01-09 by withgallantry