Helder1981/OneForAll #11: Can this run on a Raspberry PI Zero?

nhenry1647, 2020-06-28:
I love the fact that this has so many features but I am having lots of trouble getting this to run on a Pi Zero.  I have installed this on a PI3 with no issue.

Thanks!

gambino37, 2020-06-28:
The setup is no different for a pi zero

nhenry1647, 2020-06-28:
I know there is no different, but the program refuses to run.  This is what I get on multiple Zero setup attempts, even latest level of RetroPie.

Traceback (most recent call last):
  File "monitor_evdev.py", line 266, in <module>
  File "subprocess.py", line 390, in __init__
  File "subprocess.py", line 1000, in _execute_child
  File "subprocess.py", line 121, in _eintr_retry_call
KeyboardInterrupt
[791] Failed to execute script monitor_evdev
ERROR:root:ERROR: Failed start OSD binary
Traceback (most recent call last):
  File "monitor_evdev.py", line 266, in <module>
  File "subprocess.py", line 390, in __init__
  File "subprocess.py", line 1008, in _execute_child
  File "subprocess.py", line 902, in _close_in_parent
OSError: [Errno 9] Bad file descriptor
Traceback (most recent call last):
Exception AttributeError: "'  File "monitor_evdev.py", line 231, in handle_button
AttributeErrorNoneType' object has no : attribute 'suinput_destroy'"

Helder1981, 2020-06-28:
This one here is outdated and does not work on the newer Retropie, you need to head over to https://github.com/withgallantry/OneForAll where it's updated but it needs a bunch of things to be installed and run for it to work. Best bet is use the Retropei 4.51 version or wait till there is an updated binary file which might be a month away since we are all busy on other things to get back to this right now.

nhenry1647, 2020-06-28:
Thank you Helder1981 for the quick response.  I will do what you suggest.  Looking forward to buying a bunch more of these when there's stock.  Cheers!